collin80/esp32_can #1: Some examples needed

maoiot, 2018-02-27:
Great project!

It would be useful to add a few examples of the use of this library.

Really looking forward to!

reaper7, 2018-03-13:
true, library looks promising.

How to set pins for can?

collin80, 2018-03-14:
The WIP branch has a newer version of the library. In that version there's support for the MCP2515 and MCP2517FD modules as well. But, you can just plain not use those other modules if you don't want. esp32_can.cpp has the declaration where the pins are set up. And, the WIP branch has examples for how to use the library too. So, hopefully this helps.

reaper7, 2018-03-14:
pins setup as hardcoded or I can change this without editing esp32_can.cpp?

collin80, 2018-03-19:
As of the most recent commit it is now possible to change the RX and TX pins from your sketch without having to edit the library files. There's a new example that does this. You call CAN0.setCANPins(rx, tx);

ferasterkawi, 2018-11-09:
Thanks a lot Mr collin80, I faced this problem:
fatal error: freertos/FreeRTOS.h: No such file or directory
I tried to download this library from: https://codeload.github.com/maniacbug/FreeRTOS/
but I get a lot of this kind of problem like: avr/io.h no such file or directory!!!
Thanks indeed #^_^#

collin80, 2018-11-12:
It sounds like you aren't compiling for the right board type in the Arduino IDE. FreeRTOS.h is part of the ESP32 IDF so it would be available if you were compiling for the ESP32 properly. Something must be messed up somewhere. Double check you've got arduino-esp32 and that the IDE can find it properly.

collin80, 2019-01-29:
Hmm, it seems like it is trying to initialize the MCP2517FD chip three times and it fails the third time?! That doesn't seem right. Are you using CANTestESP32_FDMode unmodified?

I will do further testing to ensure that the received messages are proper. I'm not sure whether transmission might be faulty or reception.

ferasterkawi, 2019-01-29:
Without any modification I faced similar problem. 
The only modification I made is repeat sending txframe from one node!
Thanks a lot #^_^#

BPLD, 2019-10-15:
> 
> 
> A lot of thanks for u Mr. Collin #^_^#
> I'm trying to send data between tow nodes via CAN FD... So I tried CANTestESP32_FDMode file but I faced these problems:
> first, sometimes serial monitor print:
> 
> Initializing ...
> MCP2517FD SPI Inited
> _initFD()
> commonInit()
> 1100010010011110000010000000000
> 10000000000100111
> Mode
> 10101
> 0
> 38030000
> MCP2517FD SPI Inited
> _initFD()
> commonInit()
> 11100010010011110000010000000000
> 10000000000100111
> Mode
> 10101
> 0
> 38030000
> MCP2517FD Init Failed ...
> Ready ...!
> 
> second, the node sometimes receive 0 length frames, sometimes wrong data (not sent from the other node ), serial print like this:
> 
> ID: 103
> Extended: No
> Length: 0
> 
> ID: 13B0115
> Extended: Yes
> Length: 5
> 1 3E EC B7 0
> 
> ID: 39
> Extended: Yes
> Length: 8
> 0 0 0 0 4A B4 FF 8A

I was getting these odd-ball bits in my serial prints as well until I noticed you need to add the "can_common" lib to Sketch as well. FYI Collin, this works like a charm. Thank you so much!