collin80/esp32_can #8: ESP32RET CAN1 TX Queue overflow

nkirkby, 2019-05-17:
Hi Collin,
I'm working with Phil Sadow to use an ESP32RET to control some vehicle components.
I'm sending a periodic CAN message at 100Hz, and a few more at 10Hz and 1Hz (six messages in total).  This is done using either esp_timers or freeRTOS threads that call `CAN1.sendFrame()` at intervals.

I observe that after a few minutes, the transmitted frames cease to be transmitted.  The behavior isn't terribly deterministic, but I've managed to trace it back to a TX queue overflow in the MCP2517 driver by adding a print to `handleTXFifo`:

```cpp
void MCP2517FD::handleTXFifo(int fifo, CAN_FRAME_FD &newFrame)
{
  uint32_t status;
  uint16_t addr;
  BaseType_t xHigherPriorityTaskWoken = pdFALSE;

  if (fifo < 0) return;
  if (fifo > 2) return;

    //try to queue, do not wait if we can't. If we can then pretend an interrupt happened.
    if (!txQueue[fifo])
      return;
    if (xQueueSend(txQueue[fifo], &newFrame, 0) == pdPASS)
    {
      xHigherPriorityTaskWoken = xTaskNotifyGive(intTaskFD); //send notice to the handler task that it can do the SPI transaction now
    }
    else
    {
      Serial.println("Queue is full");
    }
}
```

Here is a section of the output:
```
Transmitted 13800 frames.
Transmitted 13900 frames.
Transmitted 14000 frames.
Transmitted 14100 frames.
Queue is full
Queue is full
Queue is full
Queue is full
```

I'm about to try sending with CAN0 instead of CAN1 to see if the behavior is substantially different between `esp32_can_builtin` and `mcp2517fd`.

nkirkby, 2019-05-19:
I haven't found the root cause of this issue, but I did run the following stress test:
- CAN0 and CAN1 of an ESP32RET were connected to CAN0 and CAN1 of a CANDUE.
- Bus utilization on the ESP32RET was gradually advanced
- It was observed that both TX buffers on the ESP32RET hit a maximum of about 5000-7000 messages per second at a bitrate of 500Kbaud (~100% bus utilization) before reporting TX queue overflows.

This seems to indicate that a too-high message rate was not the root cause.
It's possible in the original post that bus errors prevented the ESP32RET's CAN controllers from sending frames, thereby creating a TX buffer overflow.  This is speculation, and has not been confirmed.

Closing this issue due to vagueness.

closed 2019-05-19 by nkirkby

collin80, 2019-05-20:
I have had similar experiences. It seems that sometimes it gets into a state where bus errors are reported and then the TX buffer fills up because transmit errors cause it to go off bus. I have not found a hard and fast cause for this. One possible idea would be to investigate whether the initialization of the bus for a given speed is perhaps using sub-optimal values. You have to set the bus rate with a variety of variables - clock speed, jump width, seg1, seg2, etc. I wonder if those settings are sub-optimal or if there might be some other setting done in the initialization that is causing the hardware to not be able to stay on the bus stably. But, I haven't yet found the solution. The MCP2517FD has a LOT of debugging registers. In the newer versions of the library there is a function named printDebug that outputs the state of the debugging registers on the chip. With that function I have found that bus errors start to happen and it reports errors when trying to send. So, I know sort of why it is failing to send frames but not necessarily how to fix it yet.

nkirkby, 2019-05-20:
> In the newer versions of the library there is a function named printDebug that outputs the state of the debugging registers on the chip.

I was about to write one, then saw this already in the source and added to it.  Thanks!  Using this, I correlated ceasing of transmitted frames with NTERRCNT changing from 0 to 1.

I've made a very kloodgey fix by with the following changes:
https://github.com/collin80/esp32_can/compare/master...nkirkby:buffer_overflow_investigation

TX is more consistent than before, but is still lost for ~100 ms when the driver resets the MCP2517FD due to `needMCPreset`

No merge intent until I find a cleaner solution, just sharing findings along the way.  
Thanks for commenting.  Knowing that you're aware of this is helpful and lets me know I'm probably not making a grievous user error.

nkirkby, 2019-05-21:
Hi Collin,
I took some time to understand this:
https://github.com/collin80/esp32_can/blob/master/src/mcp2517fd.cpp#L1283
by making a call graph:
[image]
If I understand correctly, the aim of handling SPI transactions through the RTOS is to prevent a hardware-interrupt-triggered, SPI-using callback from interrupting an existing SPI transaction.  Does this sound right?

collin80, 2019-05-21:
Yes, it kind of built up over time but the idea was to try to make SPI access happen only outside of the hardware interrupt. I tried to ensure it would still be fairly fast by directly chaining to the task that handles the interrupt from the interrupt handler itself but I still wonder if perhaps there was a better approach to all this.

nkirkby, 2019-05-25:
Hi Collin,
I think what you're doing with `CAN_COMMON` is fantastic.  It really shines on the ESP32 with its separate CAN hardwares.
I've [shimmed](https://github.com/nkirkby/esp32_can/blob/develop/src/mcp2517fd.cpp) Pierre Molinaro's [acan2517](https://github.com/pierremolinaro/acan2517) into a fork of esp32_can.  Although Pierre's driver doesn't yet support CAN FD, I'm finding it works pretty well.  I have not yet found any issues.

nkirkby, 2019-05-26:
I'm wrong, Pierre has a [https://github.com/pierremolinaro/acan2517FD](library that supports FD ).

collin80, 2019-05-27:
Interesting. And this library seems to work properly for you? Perhaps I can
take some pointers from that. Interestingly, it seems to work much the same
way as my library - it uses SPI transactions and a scheme to send
notifications from the true interrupt handler to really handle things
outside the interrupt. But, it seems to rely much less on tasks. Perhaps my
overuse of tasks is not contributing to stability.

nkirkby, 2019-05-28:
I struggled for a couple days trying to get interrupt-driven SPI transactions not to conflict with task-driven SPI transactions with two MCP2517FD on a single SPI.  

The first working solution I've found is to not use interrupts at all, and simply to call the `poll()` function at 1kHz from an `esp_timer`.  Utilization is only a couple percent on the 1kHz task, even at heavy bus load on both MCPs.  It's not elegant, but it's simple.

collin80, 2019-05-29:
Hmmm, well, I suppose that could work if it positively affects stability. The MCP2517FD has what amounts to a fairly large on-die buffer so I don't think it would be possible to overflow it in 1ms. Currently the hardware is set by the library to create an 18 frame buffer. One can only send around 5000 frames per second so an 18 deep buffer is around 3 times larger than should be required at 1ms polling. 

But, I have to wonder why interrupts seem to cause such stability issues on the ESP32? I've never been able to get the MCP2517FD driver to be perfectly stable under heavy bus load but yet the much less powerful built-in CAN seems to work fine. It would seem to be an issue with the SPI related code for ESP32-Arduino but I'm not sure where.

nkirkby, 2019-05-29:
My current hypothesis for what's happening is deadlock between an interrupt-driven SPI task and a lower priority RTOS task:
[image]
The lower priority task grabs the [SPI mutex](https://github.com/espressif/arduino-esp32/blob/2fd39b1aff1c1f2be60bbd9b5fbd9c88638b0fcc/cores/esp32/esp32-hal-spi.c#L722).  Before deasserting nCS, an interrupt triggers a higher priority task.  The higher priority task cannot get the mutex, and the lower priority task is stuck yielding to it until some sort of timeout expires.

Clearly I've still not gotten to the bottom of this yet.  I'm hoping my findings so far are useful.

collin80, 2019-05-29:
Hmmm.... Well, that is an interesting theory. It is very plausible. The thing about that that bugs me is that I'm not doing ANYTHING else on the SPI that the MCP2517FD is connected to and it still breaks. It appears that you found the same thing with Pierre's library. So, that's odd. There shouldn't be any lower priority RTOS task using the SPI that could possibly assert the lock - at least I can't think of a reason it would. The ESP32 does, of course, use SPI for connection to FLASH memory but it is a different SPI bus and so should not conflict. Unless, of course, there's a bug somewhere and they do. One possible thing to try might be to lower all CAN tasks to the lowest priority or to raise them to the highest priority to see if we could be getting priority based deadlocks somewhere. 

I'm somewhat intrigued by your idea to poll at 1KHz behind the scenes. If I modified things a bit I could make the TX queue long enough to not starve during each 1ms pause. I'm not worried about the 18 position RX queue filling up in 1ms. _I agree that pollng at 1khz seems like a rather gross solution but if it is stable and no frames drop then perhaps nobody needs to know. :) At some point it becomes more trouble than it is worth to keep spending time on it. So, if this solution is more stable and does not cause any frame drops either in RX or TX then maybe it's the way to go. It's simple, it's easy, it's something other people looking at the code could easily understand and follow. It really might be the way to go.