collin80/esp32_can #15: Core Panic

MagnusThome, 2019-12-29:
Hi!

Using an MCP2515 with interrupt connected to the ESP32 and also writing to EEPROM in my code I get this crash below. It depends a bit on timing but if an interrupt arrives around the time I'm writing to EEPROM it crashes. Everything else works beautifully 👍 

```
21:21:55.295 -> Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
21:21:55.330 -> Core 0 register dump:
21:21:55.330 -> PC      : 0x400d15b0  PS      : 0x00060034  A0      : 0x40084764  A1      : 0x3ffbe1b0  
21:21:55.330 -> A2      : 0x0000001a  A3      : 0x3ffc06c4  A4      : 0x00000000  A5      : 0x04000000  
21:21:55.330 -> A6      : 0x00000000  A7      : 0x13001248  A8      : 0x80081138  A9      : 0x00000001  
21:21:55.330 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc08e0  A13     : 0x00002004  
21:21:55.364 -> A14     : 0x3ffc08e0  A15     : 0xffffdffb  SAR     : 0x0000001f  EXCCAUSE: 0x00000007  
21:21:55.364 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
21:21:55.364 -> Core 0 was running in ISR context:
21:21:55.364 -> EPC1    : 0x400867f7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d15b0
21:21:55.364 -> 
21:21:55.364 -> Backtrace: 0x400d15b0:0x3ffbe1b0 0x40084761:0x3ffbe1d0 0x400867f4:0x3ffb9d80 0x40082b5f:0x3ffb9da0 0x40088379:0x3ffb9dc0
21:21:55.398 -> 
21:21:55.398 -> Rebooting...
21:21:55.398 -> ets Jun  8 2016 00:22:57
21:21:55.398 -> 
21:21:55.398 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:21:55.398 -> configsip: 0, SPIWP:0xee
21:21:55.398 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:21:55.398 -> mode:DIO, clock div:1
21:21:55.398 -> load:0x3fff0018,len:4
21:21:55.398 -> load:0x3fff001c,len:1044
21:21:55.398 -> load:0x40078000,len:8896
21:21:55.398 -> load:0x40080400,len:5816
21:21:55.431 -> entry 0x400806ac
```

Are there any possible workarounds? Doing CAN1.disable and enable before and after the EEPROM write doesn't help.

I run the ESP32 as a bridge with two CAN interfaces. The ONLY changes I've done to the library source is this, works great!

```
////////////////////////////////////////////////////////////
Edit in library:

library\esp32_can\src\esp32_can.cpp

	ESP32CAN CAN0(GPIO_NUM_16, GPIO_NUM_17);
	//MCP2517FD CAN1(5, 27);
	MCP2515 CAN1(5, 26);

library\esp32_can\src\esp32_can.h

	//extern MCP2517FD CAN1;
	extern MCP2515 CAN1;

////////////////////////////////////////////////////////////
Connect:

ESP32		MCP2515
-----		-------
GPIO5		CS
GPIO23		SI
GPIO19		SO
GPIO18		SCK
GPIO26		INT
5V		VCC
GND		GND
		
ESP32		SN65HVD230 (or similar transceiver)
-----		----------
GPIO16		CRX
GPIO17		CTX
3V3		3V3
GND		GND

////////////////////////////////////////////////////////////
```

This test code below crashes every time ___when I have an OBD2 display transmitting traffic to the ESP32___ but of course its a bit time sensitive. Seems to only crash when there is an interrupt around the time of the EEPROM write. With _my_ OBD2 display the timing hits the mark and I get a crash 100% of the time.

```
#include "esp32_can.h"
#include "EEPROM.h"

CAN_FRAME frame;

void setup() {
  EEPROM.begin(401);
	CAN1.begin(500000);
	CAN1.watchFor();
}

void loop() {

  if (CAN1.read(frame)) {
    frame.id = 0x7E8;
    frame.data.uint8[0] = 0x03;  
    frame.data.uint8[1] = 0x41;  
    frame.data.uint8[3] = 0xFF;
    CAN1.sendFrame(frame);
    EEPROM.write(0, 1);   // beware of wear of the eeprom...
    EEPROM.commit();
  }
}
```

Deivik12, 2020-03-06:
Can you try power mcp2515 with 3.3v? Esp32 gpio`s only support 3.3v. If you power mcp2515 with 5v, it should give you problems in theory. And transceiver can be powered by 3.3v or 5v, depends on model datassheet. If that doesnt help, I would advise you to use built in can controller with vd23x transceiver, its less hardware, cheaper and more reliable.

MagnusThome, 2020-03-08:
Thanks for the input! Using a level shifter doesn't help so I doubt running on 3.3V will help but SPIFFS is a good idea!!! I'll look into that :-)

Just a wild guess from my side but I was wondering if it rather has to do with some irq calls not being in IRAM?

Deivik12, 2020-03-09:
If you tested this code without eeprom and you are 100% sure that its eeprom library, problem not mcp2515 wiring, it could be that writing eeprom(flash) at very high speeds causes somehow core panic. I dont think that irq calls are your issue here

MagnusThome, 2020-03-09:
Again, thanks for your input!

- If I add a lot of delay between writes it still crashes like this.

- It only crashes when the irq pin has a input signal triggering it. No trigger on the irq pin = no crashes at all.

jbakuwel, 2020-07-10:
[This](https://github.com/Seeed-Studio/pi-hats/issues/7) might be relevant. TL;DR: There are two limitations on the usable SPI clock speed. One is a yet unknown issue with the mcp25xxfd: it doesn't properly work with SYSCLOCK / 2 speed. Based on trial and error, the SPI clock speed was reduced to 92.5% of SYSCLOCK / 2 for the Raspberry Pi 4. This might be different for the ESP32 though.

kietDubai, 2021-03-16:
Hi !!!
I got the same Problem. And use Exception Decoder ! It shows :

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400d7540  PS      : 0x00060034  A0      : 0x40084888  A1      : 0x3ffbe1b0  
A2      : 0x0000001b  A3      : 0x3ffc0980  A4      : 0x00000000  A5      : 0x08000000  
A6      : 0x00000000  A7      : 0x13001248  A8      : 0x80081148  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800885ce  A13     : 0x3ffbe180  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40086923  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d7540

Backtrace: 0x400d7540:0x3ffbe1b0 0x40084885:0x3ffbe1d0 0x40086920:0x3ffb9d80 0x40082c83:0x3ffb9da0 0x400884dd:0x3ffb9dc0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

-------------------------------------------------------------------------------------------------------------------

PC: 0x400d7540: MCP_INTHandler() at C:\Users\telem\OneDrive\Documents\Arduino\libraries\esp32_can-master\src\mcp2515.cpp line 43
EXCVADDR: 0x00000000

Decoding stack results
0x400d7540: MCP_INTHandler() at C:\Users\telem\OneDrive\Documents\Arduino\libraries\esp32_can-master\src\mcp2515.cpp line 43
0x40086920: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 82
0x40082c83: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 62
0x400884dd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

And i did try to use CAN0 instead of CAN1 .There's no problem. What's wrong with MCP_INTHandler() at C:\Users\telem\OneDrive\Documents\Arduino\libraries\esp32_can-master\src\mcp2515.cpp line 43 ????