collin80/esp32_can #16: Auto baud-rate

Deivik12, 2020-03-04:
Hi, thanks for sharing and maintaining this source code it really does the job. Currently Im using built in canbus with vd230 transceiver. I was wondering is it possible some how in easy way to add auto baud rate?

collin80, 2020-03-04:
Yes. The easiest approach to doing autobaud is this:

1. Set the CAN hardware to listen only mode.
2. Set a speed you think "might" be it
3. Wait several hundreds of milliseconds
4. See if any traffic came in
5. If it did then that CAN speed must be correct. Turn listen only mode off. You're done.
6. If nothing came in then go back to step 1 and try a different speed.

This, of course, requires that two other nodes are on the CAN bus and talking. Naturally you can't do autobaud without some traffic to analyze.

Deivik12, 2020-03-05:
Thanks for information. Everything is clear, just not the step 1 one. Ive read how I can setup listen-only mode with esp-idf, but Im not sure how it should be done in arduino with your stack. Can you let me know how to set it to listen-only mode?

collin80, 2020-03-05:
Depending on whether you're using the built-in CAN or one of the add-ons
(MCP2515, MCP2517FD):

CAN0.setListenOnlyMode(true);
or
CAN1.setListenOnlyMode(true);

Then, obviously you can turn that off with setListenOnlyMode(false);

Deivik12, 2020-03-05:
Thanks, you are the best!

Deivik12, 2020-03-09:
I have small issue, I was testing autobaud. In my application baud speed can be either 250kbps or 500kbps. Problem Im facing is that CAN0.setListenOnlyMode(true); allows any baud speed traffic thru. Simple example Im checking 500kbps baud rate can bus with 250kbps setup:
`void setup() {
  Serial.begin(115200);
  CAN0.setListenOnlyMode(true);
  CAN0.begin(250000);
  CAN0.watchFor();
}

void loop() {
  CAN_FRAME message;
  if(CAN0.read(message)) {
  Serial.println(message.id, HEX);
  }
}`
it outputs can bus traffic which is running at 500kbps, even tho Ive tried to read it with CAN0.begin(250000); because of that, I cant make autobaud to work. I think it shouldnt output canbus traffic? If I remove CAN0.setListenOnlyMode(true); it not gonna output any canbus traffic until I set proper canbus speed. I was wondering could you should full autobaud example or help with this one somehow?

Deivik12, 2020-03-10:
I did more testing, I found that if you are trying to do CAN0.setListenOnlyMode(true); while can bus speed is 500kbps, it will somehow automatically detect canbus speed, doesnt matter what speed you choose in CAN0.begin();

Also Ive found that CAN0.setListenOnlyMode(true); doesnt work on 250kbps canbus speed. example: 
` CAN0.setListenOnlyMode(true);
CAN0.begin(500000); `
It will stop canbus completely. also it will allow send canbus frames.

In conclusion setListenOnlyMode(); is buggy and doesnt work for all speeds how it shoud. Is there other way for autobaud or maybe you can check this issue sometime?

collin80, 2020-03-10:
Alright, I will look into this. It does sound like something is broken in the listen only mode.

kietDubai, 2020-06-28:
Hi ! I got the same problem when i did testing with Auto Baud. i did call Can0.beginAutoSpeed() and Can1.beginAutoSpeed(); but it did not work. And i found that the esp32_can_builtin.cpp ! Nothing inside this 
uint32_t ESP32CAN::beginAutoSpeed()
{

}
Others work perfectly ! You did a great Work ! Sir @collin80 ! Thanks so much for this libary !

collin80, 2023-05-03:
beginAutoSpeed is certainly implemented in the both the built-in driver and the MCP2517FD driver as it sits right now. Limited testing has been done, I should test them more but the code is there now.