collin80/esp32_can #17: Add support for ESP-IDF framework

codingoverdrive, 2020-08-15:
Is it possible to modify this project so that is can be used as a dependency for Arduino or ESP-IDF projects?

If it had a library.json in PlatformIO format declaring its frameworks as both Arduino and ESP-IDF, it would at least be recognized as a compatible library for either framework. 

Is there any specific code here that fundamentally relies on the Arduino framework?

collin80, 2020-08-16:
Well, I don't think it's going to be possible as-is to use it without the Arduino layer. It does use pinMode and digitalWrite for I/O and also uses the Arduino SPI library for comm to the MCP2515 and MCP2517FD. However, basically all of the Arduino related code is in those files so it's probably possible to use the built-in CAN with only ESP-IDF so long as you remove support for MCP2515 and MCP2517FD. Alternatively, it is possible to transition the library to only using ESP-IDF functionality. Then it could be used either way.

codingoverdrive, 2020-08-16:
Understood and thanks for the information.

I'll take a look at forking the project and start with just the built-in can support for ESP-IDF

closed 2023-05-03 by collin80