collin80/esp32_can #25: Interrupts not noticed when ESP32 is heavily loaded

j4m3s, 2020-10-20:
Works fine when the ESP32 is quiet, but the interrupt from the external CAN chip is not "noticed" by the ESP32 when it's heavily loaded.  In my case I'm updating LEDs.

Investigating the problem has led me to compare the MCP2515 code (that I'm using) with the MCP2517 code - and I've realised that the 2517 code doesn't use the external interrupt at all.  Instead it polls the chip every 1ms to see whether there's anything in the buffers. 

I've raised it as an issue here to see what the thinking was behind abandoning the interrupt and using polling instead.  Can the hardware interrupt not be used reliably?

collin80, 2020-10-20:
We had some reliability issues with the MCP2517FD, especially when heavily loaded. We were having crashes with the 2517 code and were trying various things to make it more stable. The 2517 has much larger buffers on-chip than the 2515 so there was no worry at all that we could fill them before the 1ms poll. So, given the circumstances it seemed easier to poll the chip at regular intervals rather than try to process everything with an interrupt. This might be harder to do on the MCP2515 as there are far fewer hardware buffers to fall back on.

j4m3s, 2020-10-21:
Very interesting - so not a problem with the library code as much as with the chip itself?  Thank you for the insight.

I think I've narrowed-down my particular problem to the fact that the FastLED library is disabling interrupts at some points (despite the "ENABLE_INTERRUPTS" flag being set).  With only two buffers the MCP2515 fills in the meantime.  So I was wrong to blame esp32_can :)

In case it's of interest/ use to anyone else, here is the **current** functional tracing that I did to help me identify differences between the three chips supported by the library:

### MCP2515
- An interrupt is attached to INT_PIN.  On pin going low, it calls 
  - `MCP_INTHandler()` which vTaskNotifyGiveFromISR (notifies) the **intDelegateTask** semaphore
- `taskMCPInt15()` is running at priority **10** on core 0 listening for the **intDelegateTask** semaphore
    - it calls `intHandler()` when notified
      - `intHandler()` reads from the MCP2515 chip and calls
      - `handleFrameDispatch()` which puts the frame in the 
        - **callbackQueueM15** queue if there is a callback registered
        - or moves things to the **rxQueue** queue if there are no callbacks registered
          - `GetRXFrame()` or `available()` all pull from the queue
- `task_MCP15()` is running at priority **3** on core 0 watching for items in the 
  - **callbackQueueM15** queue.  It calls
      - `sendCallback()` which calls 
        - the ultimate callback!

### MCP2517
- The interrupt pin is *not connected*
- `task_MCPIntFD()` is running on core 0 at priority **19**
    - Every 1ms it calls
        - `intHandler()` which reads from the chip and calls 
            - `handleFrameDispatch()` which puts things on the
                - **callbackQueueMCP** queue
                - or the **rxQueue** if there are no callbacks registered
- `task_MCPCAN()` is running on core 0 at priority **8**
    - it's watching the **callbackQueueMCP** queue and calls 
        - `sendCallback()` which calls
          - the ultimate callback!
- `task_ResetWatcher()` is running on core 0 at priority **7**
    - it checks the chip every 2 seconds and resets the chip using  
      - `resetHardware()` and `txQueueSetup()` if needed

### Internal CAN
- `CAN_WatchDog_Builtin()` is running on core 1 with priority **10**
  - it just ticks away incrementing `cyclesSinceTraffic`
- `CAN_isr()` is called from somewhere externally.  It calls
  - `CAN_read_frame()` which puts frames into the **lowLevelRXQueue**
- `task_LowLevelRX()` is running on core 1 with priority **19**
  - it's watching the **lowLevelRXQueue** queue and passes frames to
    - `processFrame()` which checks for callbacks and puts things in the 
      - **callbackQueue** queue
      - or the **rx_queue** if there are no callbacks registered
        - `rx_avail()`, `available()` and `get_rx_buff()` all pull from the queue
- `task_CAN()` is running at priority **15** (not pinned to a core) 
  - It's watching for items in the **callbackQueue**.  It calls 
      - `sendCallback()` which calls
        - the ultimate callback!

j4m3s, 2020-10-21:
Very interesting update:  The issue I was suffering from goes away when you move the two monitoring tasks [here](https://github.com/collin80/esp32_can/blob/7c3fe2dfa74f4ad1c0c06e380f59d966ee417ac8/src/mcp2515.cpp#L134) over to core 1.

closed 2020-10-21 by j4m3s

collin80, 2020-10-22:
The MCP2517 code got a lot more testing and work than the MCP2515 code. We very quickly abandon the MCP2515 so the code sat and wasn't updated. I did find that changing which core you have tasks running on could drastically affect stability. The MCP2517 code has run one one core or the other at various points. But, how it looks now is how it was most stable for me. Core 0 is used for the WiFi and Bluetooth code so if you were trying to use those then core 0 would have many interrupts for wireless comm. Core 1 is used for the Arduino code we're writing. So, moving tasks to core 1 causes the tasks to compete with the Arduino code for processor time instead of the wireless communications code in the ESP32 core.