collin80/esp32_can #32: Disable Interupt Pin on 2515.h

hulktech, 2021-12-30:
Hi i'm using due_can library, but when it comes to CAN1 MCP2515 the function of receiving can msgs not working, only if pull up the interupt pin and then live it floating.But also in this situation after a few second the esp32-wroom-32e is restarting continuously.In the other hand for sending Can msg over the CAN1 MCP2515 is working perfectly.Just for information CAN0 eps32 built in is working also perfect with new update code for espe E versions.

I will appreciate if you can help me.

ale1800, 2022-02-25:
HI, did you solve the problem? I'm triyng to receive messages from mcp2515 but I'm not able to make it work

NMSTEC, 2022-10-24:
This is still a current problem.