collin80/esp32_can #34: getting started

qubolino, 2022-01-25:
i am trying `esp32_can/examples/CANTestESP32_BuiltIn/CANTestESP32_BuiltIn.ino` with an esp32 and a transciever HDV230.

i don't see any frames sent on the bus... any hint?

qubolino, 2022-01-25:
as a comparison i am able to see frames on the bus with the minimal example from https://github.com/sandeepmistry/arduino-CAN

collin80, 2022-01-26:
Yes, I made my library fairly early and I'm not sure if anyone had standardized which pins to use for CAN back then... And well, I chose different pins than everyone else. This is the definition in esp32_can:
ESP32CAN __attribute__((weak)) CAN0(GPIO_NUM_16, GPIO_NUM_17) ;

As you can see, I used pins 16 and 17 instead of 4 and 5.

qubolino, 2022-01-26:
indeed... after changing this it works well. thanks!

closed 2022-01-26 by qubolino