collin80/esp32_can #35: library is not working with esp32-s3

pawelka, 2022-03-22:
Platform: https://github.com/espressif/arduino-esp32/tree/esp32-s3-support branch esp32-s3-support, this branch relay on IDF-4.4 where "CAN" was removed and replaced by "TWAI" (some IDF version has backward compatible). Different names and file organisation. Below is a patch that I applied. Everything compile and can be loaded into esp32-s3 chip, but no reaction on reading or writing CAN messages. I tried the TWAI samples with HAL and it's works as expected. I have no knowledge with low level CAN/TWAI API, I don't have any idea what can be wrong. Any hint welcome.  

```
diff --git a/src/esp32_can_builtin_lowlevel.cpp b/src/esp32_can_builtin_lowlevel.cpp
index 052b3ca..b50035d 100644
--- a/src/esp32_can_builtin_lowlevel.cpp
+++ b/src/esp32_can_builtin_lowlevel.cpp
@@ -29,8 +29,9 @@
 #include "freertos/FreeRTOS.h"
 #include "freertos/queue.h"
 
-#include "esp_intr.h"
+#include "esp_intr_alloc.h"
 #include "soc/dport_reg.h"
+#include "soc/periph_defs.h"
 #include <math.h>
 
 #include "driver/gpio.h"
@@ -214,20 +215,20 @@ int CAN_init()
     CANBI_ENTER_CRITICAL();
 
     //enable module
-    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
-    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
+    DPORT_SET_PERI_REG_MASK(SYSTEM_PERIP_CLK_EN0_REG, SYSTEM_TWAI_CLK_EN);
+    DPORT_CLEAR_PERI_REG_MASK(SYSTEM_PERIP_RST_EN0_REG, SYSTEM_TWAI_RST);
 
     MODULE_CAN->MOD.B.RM = 1; //first thing once module is enabled at hardware level is to make sure it is in reset
 
     //configure TX pin
     gpio_set_level(CAN_cfg.tx_pin_id, 1);
     gpio_set_direction(CAN_cfg.tx_pin_id,GPIO_MODE_OUTPUT);
-    gpio_matrix_out(CAN_cfg.tx_pin_id,CAN_TX_IDX,0,0);
+    gpio_matrix_out(CAN_cfg.tx_pin_id,TWAI_TX_IDX,0,0);
     gpio_pad_select_gpio(CAN_cfg.tx_pin_id);
 
     //configure RX pin
 	gpio_set_direction(CAN_cfg.rx_pin_id,GPIO_MODE_INPUT);
-	gpio_matrix_in(CAN_cfg.rx_pin_id,CAN_RX_IDX,0);
+	gpio_matrix_in(CAN_cfg.rx_pin_id,TWAI_RX_IDX,0);
 	gpio_pad_select_gpio(CAN_cfg.rx_pin_id);
 
     //set to PELICAN mode
@@ -276,7 +277,7 @@ int CAN_init()
     MODULE_CAN->BTR1.B.SAM = 0x1;
 
     //enable all interrupts (BUT NOT BIT 4 which has turned into a baud rate scalar!)
-    MODULE_CAN->IER.U = 0xEF; //1110 1111
+    MODULE_CAN->IER.U = 0x1EF; //1 1110 1111
 
     //no acceptance filtering, as we want to fetch all messages
     MODULE_CAN->MBX_CTRL.ACC.CODE[0] = 0xfff;
@@ -300,7 +301,7 @@ int CAN_init()
     (void)MODULE_CAN->IR.U;
 
     //install CAN ISR
-    esp_intr_alloc(ETS_CAN_INTR_SOURCE, ESP_INTR_FLAG_IRAM, CAN_isr, NULL, NULL);
+    esp_intr_alloc(ETS_TWAI_INTR_SOURCE, ESP_INTR_FLAG_IRAM, CAN_isr, NULL, NULL);
 
     //Showtime. Release Reset Mode.
     MODULE_CAN->MOD.B.RM = 0;
@@ -322,8 +323,8 @@ int CAN_stop()
 	//enter reset mode
 	MODULE_CAN->MOD.B.RM = 1;
 
-    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
-    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
+    DPORT_CLEAR_PERI_REG_MASK(SYSTEM_PERIP_CLK_EN0_REG, SYSTEM_TWAI_CLK_EN);
+    DPORT_SET_PERI_REG_MASK(SYSTEM_PERIP_RST_EN0_REG, SYSTEM_TWAI_RST);
     
     MODULE_CAN->IER.U = 0; //enable no interrupts
 
```

pawelka, 2022-03-26:
I ported it to use TWAI hal_ll abstraction to work with esp32-s3. Result is here: https://github.com/pawelka/esp32_can

closed 2022-03-26 by pawelka

Sheimy, 2022-10-25:
This is how to use TWAI with ESP32 S3
https://github.com/Sheimy/TWAI_S3_LISEN