collin80/esp32_can #36: CAN/TWAI Support for ESP32-C3 ?

heliophagus, 2022-05-08:
Would it be possible to modify this library to support the ESP32-C3 microcontroller? It has the same CAN controller (SJA1000) as the other ESP32s, but a different, RISC-V based microcontroller. The existing Arduino CAN libraries will not compile for ESP32-C3s, giving the error message "fatal error: soc/dport_reg.h: No such file or directory." The ESP32-C3 is becoming increasingly popular, and modifying this CAN/TWAI library to support it would be a HUGE help to the community.

eastwife, 2022-09-03:
Same for me, _**soc/dport_reg.h: No such file or directory**_

MagnusThome, 2022-10-10:
Look at this fork with support for S2, C3 and S3 AFAIK?

https://github.com/pawelka/esp32_can

eastwife, 2022-10-10:
> Look at this fork with support for S2, C3 and S3 AFAIK?
> 
> https://github.com/pawelka/esp32_can

Not working.

```
Alternatives for soc/dport_reg.h: []C:\Users\eastw\Documents\Arduino\libraries\esp32_can-master\src\esp32_can_builtin_lowlevel.cpp:33:10: fatal error: soc/dport_reg.h: No such file or directory

 #include "soc/dport_reg.h"ResolveLibrary(soc/dport_reg.h)

          ^~~~~~~~~~~~~~~~~
  -> candidates: []
compilation terminated.
```

Sheimy, 2022-10-25:
Try This one  i try it with S3
https://github.com/Sheimy/TWAI_S3_LISEN

kietDubai, 2022-11-13:
> Try This one i try it with S3 https://github.com/Sheimy/TWAI_S3_LISEN

It worked Bro! Thankyou

bakermat, 2023-01-13:
Does anyone have a similar fix for the C3 specifically as well?

collin80, 2023-05-03:
The library has been changed to be based upon the ESP-IDF TWAI driver so it should properly support the S3 and C3 now.

e52cdb7de35744f1601448f611b22acdc920cc2c

closed 2023-05-03 by collin80

MagnusThome, 2023-05-03:
Fantastic! Thanks for your efforts.

Email me at magnust@gmail.com with your address and I'll send you a RejsaCAN board if you want :-)

MagnusThome, 2023-05-03:
@collin80 

In what IDE should this compile? I can't get it to compile for either C3 or S3 in the Arduino IDE. I feel like I'm missing something obvious here :-)  In the board manager I have esp32 Espressif 2.0.8 installed. 

Trying to compile the CANTestESP32_BuiltIn.ino example for "ESP32C3 Dev Module" I get a short error notice:

```
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:33:10: fatal error: soc/dport_reg.h: No such file or directory
 #include "soc/dport_reg.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32C3 Dev Module.
```

And trying to compile the CANTestESP32_BuiltIn.ino example for "ESP32S3 Dev Module" I get a long error like this:

```
In file included from D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:32:
C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:38:
D:\Documents\-  ARDUINO\libraries\esp32_can\src\can_regdef.h:209:10: error: 'CAN_FIR_t' does not name a type; did you mean 'CAN_FRAME'?
          CAN_FIR_t FIR;      /**< \brief Frame information record */
          ^~~~~~~~~
          CAN_FRAME
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp: In function 'BaseType_t CAN_read_frame()':
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:120:45: error: 'volatile struct CAN_Module_t::<unnamed union>::<unnamed>' has no member named 'FIR'
  __frame.FIR.U = MODULE_CAN->MBX_CTRL.FCTRL.FIR.U;
                                             ^~~
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp: In function 'int CAN_write_frame(const CAN_frame_t*)':
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:176:29: error: 'volatile struct CAN_Module_t::<unnamed union>::<unnamed>' has no member named 'FIR'
  MODULE_CAN->MBX_CTRL.FCTRL.FIR.U=p_frame->FIR.U;
                             ^~~
In file included from C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_reg.h:21,
                 from D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:33:
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp: In function 'int CAN_init()':
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:217:29: error: 'DPORT_PERIP_CLK_EN_REG' was not declared in this scope
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                             ^~~~~~~~~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:217:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:217:29: note: suggested alternative: 'SYSTEM_PERIP_CLK_EN1_REG'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                             ^~~~~~~~~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:217:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
     ^~~~~~~~~~~~~~~~~~~~~~~
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:217:53: error: 'DPORT_CAN_CLK_EN' was not declared in this scope
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                                                     ^~~~~~~~~~~~~~~~
[... cut ...]
C:\Program Files (x86)\Arduino\portable\packages\esp32\hardware\esp32\2.0.8/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
D:\Documents\-  ARDUINO\libraries\esp32_can\src\esp32_can_builtin_lowlevel.cpp:326:5: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
     ^~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32S3 Dev Module.
```

collin80, 2023-05-03:
The file causing the compile issues (esp32_can_builtin_lowlevel.cpp) should no longer exist. If you check the repo you'll see that I deleted it. A bunch of files were no longer needed and in fact will cause compile errors on anything other than the original ESP32. I don't know why git didn't automatically remove them for you. Maybe it doesn't remove files by default even if they're removed from the repo. You may need to delete your copy of the library and redownload it to make sure you don't have any of the offending old files that cause problems.

MagnusThome, 2023-05-03:
Thx! Worked like a charm now 👍 

And again, email me at magnust@gmail.com with your address and I'll be happy to send you a RejsaCAN ESP32-S3 board if you want :-)