collin80/esp32_can #39: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead

JimNickerson, 2022-08-11:
In file included from D:\Documents\ARDUINO\libraries\esp32_can-master\src\esp32_can_builtin_lowlevel.cpp:32:

...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]

collin80, 2022-08-12:
Yes, I've seen the same warning. It really ought to be fixed so I'm assigning it to myself so I remember to correct it the next time I look at the code.

JimNickerson, 2022-08-12:
Thank you for looking.
I copied the library to my local Arduino/Libraries folder
I changed src/esp32_can_builtin_lowlevel.cpp line 12 to
"#include "esp_intr_alloc.h" // jimn warning depreciated #include "esp_intr.h""
I bumped the libary.properties  version by .1 so Arduino would resolve to my changed version
My code compiles now without the warning.
Tomorrow I hope to try my app on a real system to confirm it affects nothing else.

JimNickerson, 2022-10-10:
Tomorrow came and went and I am just now getting to responding.
I can confirm my changes work well if using esp32 chip
This library will not work on the other variants like -S3 etc.
The esp32_can_builtin_lowlevel.cpp relies on register definitions from files that are not available in for variants like -S3.
This esp32_can_builtin_lowlevel.cpp code is quite a clever bit of work to adapt to the esp32 can peripheral
After some work and thought I had a look at vscode/esp-idf and the provided example code.
The twai_network_master and twai_network_slave worked very well on the esp32-s3
I adapted them to an Arduino ino and they worked well.
I am thinking about the possibility of converting the twai code for use in this library instead of esp32_can_builtin_lowlevel., I like using this collin80 library

collin80, 2022-10-11:
Yes, this may be a good idea. The problem here is that my library was written before there was any official CAN library. The esp32_can_builtin_lowlevel code is from someone else who gathered together all the required headers, definitions, etc before Espressif ever did anything themselves. So, I really appreciate that someone did this and I was able to build on top and make it work. But, these days there is a stable CAN/TWAI library which is official and probably better designed and compatible with more of their chips. So, it certainly makes sense to replace the current low level code with the official TWAI library and just build on top of it. I just haven't had a chance to look into that.

Sheimy, 2022-10-25:
This is how to use TWAI with ESP32 S3
https://github.com/Sheimy/TWAI_S3_LISEN

collin80, 2023-05-02:
I'm probably around 80% of the way toward converting this library to use the ESP-IDF TWAI driver. It works for sending right now but not for receiving. This is likely an easy bug but I don't have the time yet today to finish it up. Still, within a day or two the library will be fully based around TWAI when it comes to the built-in CAN on the various ESP32 chips so S3 will then be supported.

collin80, 2023-05-03:
It's done (in theory). I've tested the CAN to work for sending and receiving but more testing is always useful. And, I don't have an ESP32-S3 yet so I haven't tested it to work on that chip but it should.

e52cdb7de35744f1601448f611b22acdc920cc2c