collin80/esp32_can #40: HELLO, i tried esp32 with mcp2515 bun not working

lepromer, 2022-08-14:
HELLO, i tried esp32 with mcp2515 bun not working, did you tested with esp32, what i can i doing wrong,
CS 5
INT 27
CLK 18 
SI 19 
SO 23

collin80, 2022-08-14:
I'm not sure. The MCP2515 code is very old at this point. Fairly quickly we moved to the CAN-FD hardware and the 2515 code was not touched since. You will probably have to do some debugging by dumping registers and/or putting debugging statements at various places to see what is going on.

AsmodeusX, 2022-08-15:
I can confirm, it doesn't initialize but won't work at all.

jeroenveer, 2022-08-16:
Works fine on my ESP32 with MCP25625 did needed some minor changes.

NMSTEC, 2022-10-25:
This seems to be a SPI collision, still working on it. 
[image]

NMSTEC, 2022-10-25:
[Running while transmitting.txt](https://github.com/collin80/esp32_can/files/9864459/Running.while.transmitting.txt)

NMSTEC, 2022-10-26:
Seems we are never clearing the interrupt flags. So it stops interrupting as per docs.
[image]
 Thats issue 1,

Issue 2 is everytime the MCP2515 stops transmitting it, we for some reason request to read "mode_normal"... 
[image]

This has got to be a simple issue somwhere, and yet im still fighting it. half a mind to start rewriting library myself step by step. Currently on vacation so no JTAG debugging for me :(