collin80/esp32_can #44: Add support for low bitrates

bakermat, 2023-01-13:
I'd like to use the ESP32's CAN bus functionality to monitor my heatpump, which uses a bitrate of 20KBPS. I see it's currently not supported, would it be possible to add this (and perhaps other common low bitrates)?

collin80, 2023-05-03:
I will add it. However, only newer ESP32 chips can do it. The ESP32 uses an 80MHz clock and the divisor only went up to 128 on older chips. 80MHz / 128 = 625k. Then divide that by your desired speed and you get 625k / 20k = 31.25. It's complicated but the max number acceptable there is 25 so you just can't get there. 25k is the absolute slowest speed possible on all ESP32 boards. So, to achieve 20k speed you will need a rev2 ESP32 or an ESP32-S3 based board.

collin80, 2023-05-03:
Here is the commit where 20k was added: e52cdb7de35744f1601448f611b22acdc920cc2c

closed 2023-05-03 by collin80