collin80/esp32_can #46: sending multiple CAN frames

weeblebiker, 2023-04-16:
Hi,
I'm attempting to send 12 PID request can frames to my car's bus

`void CAN() {
  
  static unsigned long l = 0;
  unsigned long t = millis();
  if ((t - l) > 1000) {
    sendPIDRequest(0x7DF, 0x06);  //STFT
    sendPIDRequest(0x7DF, 0x07);  //LTFT
    sendPIDRequest(0x7DF, 0x0E);  //TA
    sendCustomRequest(0x720, 0x2A, 0x05);
    sendCustomRequest(0x720, 0x2A, 0x0A);
    sendCustomRequest(0x720, 0x2A, 0x06);
    sendCustomRequest(0x720, 0x2A, 0x0B);
    sendCustomRequest(0x720, 0x2A, 0x07);
    sendCustomRequest(0x720, 0x2A, 0x0C);
    sendCustomRequest(0x720, 0x2A, 0x08);
    sendCustomRequest(0x720, 0x2A, 0x0D);

    //sendCustomRequest(0x720, 0x03, 0xEC); //KR
    l = t;
  }`
does the library stagger the transmissions or is my code spamming the bus with 12 pid requests back to back with no  time for response between? the requests at the bottom of the list sometimes take minutes to get response. I'm considering spacing all the requests 100ms apart, if the library isn't spacing the requests.

collin80, 2023-05-01:
The library doesn't add any spacing. It will try to send messages as fast as it can if you queue them up. This probably isn't what you are looking to do. Yes, spacing the requests by about 100ms would be a decent idea. Or, only send a new request after the last one has replied (or after some longer delay to handle the case where the request was lost).

weeblebiker, 2023-05-02:
I can confirm adding 75+ms between requests fixes the behavior. I will see if I can cobble something together that sends next after reply is received which is a better behavior rather than my initial spam, then time based sequences.

closed 2023-05-03 by collin80