collin80/esp32_can #47: ESP crashes when you add attachInterrupt

jeroenveer, 2023-05-01:
When you add a GPIO interrupt it keeps crashing.
Without the CAN library it runs fine and without attachInterrupt it also runs fine.

```
#include <esp32_can.h>

int SpeedCounter = 0;

void ARDUINO_ISR_ATTR ISR_SPEEDCOUNT() {
	SpeedCounter++;
}

void setup() {

	pinMode(22, INPUT_PULLUP);
	attachInterrupt(22, ISR_SPEEDCOUNT, FALLING);
}

void loop() {
  
}
```

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
E (134) gpio: esp_ipc_call_blocking failed (0x103)
[     2][E][esp32-hal-gpio.c:175] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start
E (137) esp_cor��fVW}���͡� No core dump partition found!
E (141) esp_core_dump_flash: No core dump partition found!
[    17][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (38) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f165b  PS      : 0x00060e33  A0      : 0x800d5907  A1      : 0x3ffb21d0  
A2      : 0x00000000  A3      : 0x3ffc3040  A4      : 0x3ffc3044  A5      : 0x00060e23  
A6      : 0x00060e20  A7      : 0x00000001  A8      : 0x800d5550  A9      : 0x3ffb2190  
A10     : 0x3ffbdbf0  A11     : 0x00400000  A12

collin80, 2023-05-01:
Hmm, apparently I hadn't ever tried this. I suspect it may have something to do with this line in your output:
E (38) gpio: gpio_install_isr_service(449): GPIO isr service already installed

When I get a chance I'll try to get to the bottom of this.

jeroenveer, 2023-05-02:
I have disabled this line
`MCP2515 __attribute__((weak)) CAN1(15, 33) ;`

only using the internal CAN on this board, on another board i use both internal and external mcp25625.
So will enable/disable this in the library when i switch boards.

jeroenveer, 2023-05-02:
Tested the MCP25625 and can transmit is working but it would not receive, it only received when there was a msg transmitted.
The interrupt was not working giving the error:
E (134) gpio: esp_ipc_call_blocking failed (0x103)
[ 2][E][esp32-hal-gpio.c:175] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start

I disabled the attachInterrupt in mcp2515.cpp
```
MCP2515::MCP2515(uint8_t CS_Pin, uint8_t INT_Pin) : CAN_COMMON(6) {
  pinMode(CS_Pin, OUTPUT);
  digitalWrite(CS_Pin,HIGH);
  //pinMode(INT_Pin,INPUT);
  //digitalWrite(INT_Pin,HIGH);
  
  //attachInterrupt(INT_Pin, MCP_INTHandler, FALLING);
```

and added IRAM_ATTR:
```
void IRAM_ATTR MCP_INTHandler() {
  BaseType_t xHigherPriorityTaskWoken = pdFALSE;
  vTaskNotifyGiveFromISR(intDelegateTask, &xHigherPriorityTaskWoken); //send notice to the handler task that it can do the SPI transaction now
  if (xHigherPriorityTaskWoken == pdTRUE) portYIELD_FROM_ISR(); //if vTaskNotify will wake the task (and it should) then yield directly to that task now
}
```

and set the INT pin in setup:
```
	CAN1.setINTPin(33);
	if (CAN1.begin(500000))
	{
		Serial.println("MCP2517FD Init OK ...");
	}
	else {
		Serial.println("MCP2517FD Init Failed ...");
	}
	CAN1.watchFor(); //allow anything through
```

Receiving is now working with interrupts.
And adding additional GPIO interrupt is now also working.

jeroenveer, 2023-05-02:
This also works instead of in setup
```
bool MCP2515::_init(uint32_t CAN_Bus_Speed, uint8_t Freq, uint8_t SJW, bool autoBaud) {
    setINTPin(_INT);
```

jeroenveer, 2023-05-04:
Just make the changes discriped above in your library.