collin80/esp32_can #48: assert failed: spinlock_acquire spinlock.h:122

MrDIYca, 2023-05-09:
I am getting spinlock error when using Arduino ESP32 Espressfi version 2.0.7 or higher (2.0.7, 2.0.8 and 2.0.9). 

2..06 and lower work perfectly fine.

Using the sample code below results is constant rebooting at CAN0.begin

```
#include <esp32_can.h>
#define  SHIELD_LED_PIN   26

void setup(){
  Serial.begin(115200);
  CAN0.setCANPins(GPIO_NUM_5, GPIO_NUM_4);
  CAN0.setListenOnlyMode(true);
  CAN0.begin(500000); // 500Kbps
  CAN0.watchFor();
}

void loop(){
  CAN_FRAME can_message;

  if (CAN0.read(can_message))  {
    Serial.print("CAN MSG: 0x");
    Serial.print(can_message.id, HEX);
    Serial.print(" [");
    Serial.print(can_message.length, DEC);
    Serial.print("] <");
    for (int i = 0; i < can_message.length; i++)    {
      if (i != 0) Serial.print(":");
      Serial.print(can_message.data.byte[i], HEX);
    }
    Serial.println(">");
  }
}
```

Serial output:  
`assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE) Backtrace: 0x400833dd:0x3ffb5060 0x40087f2d:0x3ffb5080 0x4008d281:0x3ffb50a0 0x4008ac8b:0x3ffb51d0 0x40088c0e:0x3ffb5210 0x400d726f:0x3ffb5250 0x400d1b93:0x3ffb5290 ELF file SHA256: 6165a1cc5a50c39b Rebooting...`

Board: ESP32 DevKit 1 Espressif - version 2.0.6
IDE: Arduino
CAN: SN65HVD233
OS: MacOS

collin80, 2023-05-10:
I installed 2.0.9 and tried it. I also get the same result. Here is the decoded stack dump:

0x400833dd: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x40087f2d: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x4008d281: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x4008ac8b: spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h line 122
:  (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c line 301
0x40088c0e: vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h line 578
:  (inlined by) xQueueReceive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 1400
0x400d725b: twai_receive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/twai.c line 595
0x400d1b7f: task_LowLevelRX(void*) at /home/collin/Arduino/libraries/esp32_can/src/esp32_can_builtin.cpp line 97

So, I'll look into what is going on and fix it.

collin80, 2023-05-10:
Should be fixed here: 678480fe8f4a9b59205920deac58583b0b8d83ae

It seems there was a race condition between a task I started and the TWAI library. It seems to need some delay between stopping the TWAI driver and uninstalling it. Adding some delay seems to have fixed it for me. I tested it to successfully send and receive.

iboldurev, 2023-05-29:
I have the same error, the problem is not solved

iboldurev, 2023-05-29:
I use platformio and have a same error

HonzaCZ, 2023-07-02:
Hi there,

I've just started using your library in my project. I like that it has a built-in mechanism for callback invocation on a specific filter, which is not common among many libraries.

However, here's the issue: Everything works perfectly until I start the device on an active bus with lot of traffic (500 kbps, about 1200 msg/sec). At that point, I receive a bunch of error messages

`E (5318) TWAI: Alert 2048`

and it usually ends up in a reboot due to an error.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008b1db  PS      : 0x00060033  A0      : 0x8008a189  A1      : 0x3ffbf2ec  
A2      : 0x0000001d  A3      : 0x3ffc38fc  A4      : 0x00000004  A5      : 0x00060023  
A6      : 0x00060023  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000001  
A10     : 0x00000003  A11     : 0x00000003  A12     : 0x3ffb5200  A13     : 0x3ffb51d0  
A14     : 0x007bf3b8  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x400864b1  LEND    : 0x400864d3  LCOUNT  : 0xffffffff  

Backtrace: 0x4008b1d8:0x3ffbf2ec |<-CORRUPTED
```

or

```
assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x40083b01:0x3ffb5060 0x400886a1:0x3ffb5080 0x4008da15:0x3ffb50a0 0x4008b40b:0x3ffb51d0 0x40089382:0x3ffb5210 0x400d8efb:0x3ffb5250 0x400d4cb7:0x3ffb5290
```

It's likely related to the disable() and enable() method and their invocation in set_baudrate(). When I have the driver properly configured from a previous run, and I comment out the set_baudrate(ul_baudrate); call in init(), the problem doesn't occur. Even when I turn on the device on a very busy bus , it starts receiving messages normally.

From what I've researched and tested, the issue might be related to the incoming message buffer overflow when the twai driver is disabled and enabled set_baudrate().

I can provide further details and reliably simulate the problem if needed.

Please let me know how I can assist you further.

HonzaCZ, 2023-07-03:
Hi again. After more testing I'm pretty sure errors are caused by call of twai_driver_uninstall() in disable() method when there are still exist library tasks (task_CAN, task_LowLevelRX, CAN_WatchDog_Builtin). Waiting here does not help. Tried 10 sec and result same but 10 sec later. 😁 But it helps to delete tasks after twai_stop() and before twai_driver_uninstall(), so with queues.

I tried to solve the problem and digged a bit deeper. In the end, I managed to resolve it to my satisfaction. I'm sending a PR, let me know what you think about it.

commit 37711116 (2024-06-11): Fixed #48 Better handling of tasks and queues when enabling and disabling TWAI driver
commit 66d1440c (2024-06-11): Merge pull request #50 from HonzaCZ/master

Fixed #48 Better handling of tasks and queues when enabling and disabling TWAI driver
closed 2024-06-11 by collin80