collin80/esp32_can #52: Hanging in task_LowLevelRX

lordryck, 2023-08-17:
I've been using this library on one of those Fusion CAN32 boards EVTV used to sell, quite successfully. I had been using Arduino IDE 1.8.13. Two days ago I upgraded to IDE 2.1.1 which, among other things, upgraded the compiler and system libraries. My functioning code started crashing with a null pointer exception in can_common.cpp. I was behind there too so I downloaded the latest versions of can_common and esp32_can. The crash went away but now it hangs. I don't have a jtag debugger but good old print statements show that when ESP32CAN::_init in esp32_can_builtin.cpp registers task_LowLevelRX the call to  xTaskCreatePinnedToCore does not return. Putting a print in the while loop in task_LowleveRX shows the loop is running so the board isn't hung, it's just not continuing. Any ideas?

collin80, 2023-08-17:
Hmmm, I'm not really sure what is going on in your case. I had TERRIBLE trouble at first getting the ESP32_CAN library to work with the newer ESP32-S3 chips. It would do -pretty much what you are describing - hang in strange places but seem like the processor was still running, it just was hung in an infinite loop somewhere. I had to make a lot of little changes to make it work. Now I've been using the library on both ESP32 and ESP32-S3 boards and it seems to operate properly. Did you completely delete your old versions of the can_common and esp32_can libraries and then redownload them? The structure of the esp32_can library changed quite a bit recently because it now uses the official espressif CAN driver.

lordryck, 2023-08-18:
Yes, I deleted both directory trees then installed the newest ones from here. I noticed you were using the TWAI drivers now. I also deleted the board from the IDE and reinstalled it thinking it might be something lingering there.

I just changed that call from xTaskCreatePinnedToCore to a just xTaskCreate and got `<<<<<<E (11837) TWAI: Alert 4096` followed by a never ending line of capital T's. Dunno if that helps at all

lordryck, 2023-08-20:
Have you tried the latest Arduino UI? 2.1.1? I only ask because that's what appears to have broken my systems and I just tried using an SKPang ESP32 board and it exhibits the same behavior. I don't want to recompile my ESP32 CANDue systems for fear they'll break too. I also upgraded the firmware on the original fusion to Espressif's latest 3.2.0.0 with no change. My gut feeling is that something in the compiler upgrade is the actual culprit. Like some data structure moved to align on a different boundary or something. I'm going to take the plunge and try them on ESP-IDF or Platform I/O

collin80, 2023-08-21:
I do have the 2.1.1 version of the IDE on my computer. I almost always use 1.8.19 just because I like to use an external editor (VSCode) and I'm used to the 1.8 IDE so I just stick with it generally. But, I'll try to compile with 2.1.1 and see if that changes anything.

collin80, 2023-08-22:
I have compiled esp32ret with Arduino 2.1.1 and arduino-esp32 2.0.11 and it seems to be working fine for me.  I'm up over 150 seconds with CAN traffic coming in and no resets or crashes. One thing to check, I used pins 16 and 17 on the newer EVTV boards for CAN. A lot of other boards use pins 4 and 5. So, make sure it is using the proper pins for your hardware. This CAN library does have the ability to change the pin assignment. Here is an example line found in esp32ret: CAN0.setCANPins(GPIO_NUM_4, GPIO_NUM_5);

lordryck, 2023-08-22:
I just love it when I have totally weird problems. The Fusion32 boards use 4 & 5. The SKPang board I have uses 25 and 26. Both were working perfectly until I upgraded the IDE. FWIW, I always use an external editor. The old IDE, of course, has the checkbox. 2.1.1 does it right--save the file in the external editor and the IDE instantly refreshes its copy. My editor does too, so I can edit it both places if I want to.

It is a really bizarre situation. The call to xTaskCreatePinnedToCore just never returns. Today I installed installed Platformio into VSCode and ported the program to that. It's not hanging. Instead it continuously crashes with the below. Still pointing to something on core 1

ELF file SHA256: ae0d43cffdfca74b

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x4014515b  PS      : 0x00060236  A0      : 0x8014b641  A1      : 0x3ffb00a0  
A2      : 0x3ffb81e8  A3      : 0x3ffb03b0  A4      : 0x3ffb059b  A5      : 0x3ffb0540  
A6      : 0x3ffb0480  A7      : 0x00000004  A8      : 0x8008d794  A9      : 0x3ffb03f0  
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x3ffb09a0  
A14     : 0x3ffb08e0  A15     : 0x00000004  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40089839  LEND    : 0x40089849  LCOUNT  : 0xffffffff  

Backtrace: 0x40145158:0x3ffb00a0 0x4014b63e:0x3ffb03b0 0x4014b67a:0x3ffb0440 0x400d2d86:0x3ffb0480 0x400d2d8c:0x3ffb0560 0x400d2d8c:0x3ffb0640 0x400d2d8c:0x3ffb0720 0x400d2d8c:0x3ffb0800 0x400d2d8c:0x3ffb08e0 0x400d2d8c:0x3ffb09c0 0x400d2d8c:0x3ffb0aa0 0x400d2d8c:0x3ffb0b80 0x400d2d8c:0x3ffb0c60 0x400d2d8c:0x3ffb0d40 0x400d2d8c:0x3ffb0e20 0x400d2d8c:0x3ffb0f00 0x400d2d8c:0x3ffb0fe0 0x400d2d8c:0x3ffb10c0 0x400d2d8c:0x3ffb11a0 0x400d2d8c:0x3ffb1280 0x400d2d8c:0x3ffb1360 0x400d2d8c:0x3ffb1440 0x400d2d8c:0x3ffb1520 0x400d2d8c:0x3ffb1600 0x400d2d8c:0x3ffb16e0 0x400d2d8c:0x3ffb17c0 0x400d2d8c:0x3ffb18a0 0x400d2d8c:0x3ffb1980 0x400d2d8c:0x3ffb1a60 0x400d2d8c:0x3ffb1b40 0x400d2d8c:0x3ffb1c20 0x400d2d8c:0x3ffb1d00 0x400d2d8c:0x3ffb1de0 0x400d2d8c:0x3ffb1ec0 0x400d2d8c:0x3ffb1fa0 0x400d2d8c:0x3ffb2080 0x400d2d8c:0x3ffb2160 0x400d335f:0x3ffb2240 0x400dbfe2:0x3ffb2290

lordryck, 2023-08-26:
The above was caused by me trying to print Wifi.localIP() which, it turns out, isn't a string. I changed it to Wifi.localIP().toString() and that crash went away. Today I fixed the hang in esp32_can by putting vTaskDelay(1) inside the loop in task_LowLevelRX. I'm not sure why it's necessary but it seems even a one tick delay lets the scheduler take over and do other stuff.

AlbertoEusebio, 2023-08-28:
I had the same problem with an ESP32. I think that task_LowLevelRX is starving due to the scheduler keeps scheduling the other task on  CORE1(the watchdog). As suggested by @lordryck, adding `vTaskDelay(1)` solves the problem with the scheduler.
Anyway, this solution is not nice, because it is impossible to predict whether it will crash or not on slightly different systems and codes, so a solution using semaphores or task yelding would be more appropriate but unfortunately, all my attempts were unsuccessful.