collin80/esp32_can #56: beginAutoSpeed Crashes ESP32

bl1tz3n4, 2023-09-30:
Been having a difficult time getting auto speed to work.. When I run it and it encounters the speed of the data, it crashes citing a core 1 panic. Below is the feed from my serial terminal:

```
Trying Speed 1000000 FAILED.
Trying Speed 500000Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008ac77  PS      : 0x00060033  A0      : 0x80089c25  A1      : 0x3ffbf2ac  
A2      : 0x0000001f  A3      : 0x3ffc2ecc  A4      : 0x00000004  A5      : 0x00060023  
A6      : 0x00060023  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000001  
A10     : 0x00000005  A11     : 0x00000003  A12     : 0x00000000  A13     : 0x00000020  
A14     : 0x007bf378  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000023  LBEG    : 0x40085d55  LEND    : 0x40085d77  LCOUNT  : 0xffffffff  

Backtrace: 0x4008ac74:0x3ffbf2ac |<-CORRUPTED

ELF file SHA256: efd7b5ffc96f16e9
```

I believe I am using the function correctly, but I did not see any examples to confirm or deny this. 

Code is:

```
ESP32CAN CAN0(GPIO_NUM_4, GPIO_NUM_5);

void setup() {
  Serial.begin(115200);
  delay(1000);
  CAN0.begin(250000);
  CAN0.watchFor();
  CAN0.setListenOnlyMode(true);
  CAN0.beginAutoSpeed();
}
```

Is there something I am missing? I am using IDE v2.1.1 with the latest versions from here.

collin80, 2023-10-02:
Try not beginning CAN0 with a definite speed first. beginAutoSpeed() does the initialization process. It also should automatically be using listen only mode internally so start with beginAutoSpeed then do CAN0.watchFor() afterward and maybe CAN0.setListenOnlyMode(true) if you really don't want to touch the bus.

kevinlieb, 2024-04-02:
@collin80 can you perhaps post an example of "beginAutoSpeed()" usage?
and how to retrieve what rate it discovered?