collin80/esp32_can #57: ESP32 crash or wont initialize

notsolowki, 2023-10-05:
When i try to run the sketch ESP32_builtin, if i dont set it to listen only mode i wont receive anything. If i use normal mode i have to change the .cpp and add  vTaskDelay(1); as others mentioned. However if i add  vTaskDelay(1); the code no longer hangs in normal mode at begin() but it constantly bootloops after it tries to tx a message. What can i do to fix this?

`void task_LowLevelRX(void *pvParameters)
{

    ESP32CAN* espCan = (ESP32CAN*)pvParameters;
    while (1)
    {
	 vTaskDelay(1);
        twai_message_t message;
        if (twai_receive(&message, pdMS_TO_TICKS(100)) == ESP_OK)
        {
            espCan->processFrame(message);
        }
    }
}
`

notsolowki, 2023-10-05:
What does this error mean? TWAI: Alert 4096

collin80, 2023-10-06:
TWAI alert 4096 means:
#define TWAI_ALERT_ERR_PASS                 0x00001000  /**< Alert(4096): TWAI controller has become error passive */

Which suggests that it doesn't think things are right with the CAN bus. 

I'm not sure why it is boot looping. It should give you a stack trace when that happens. You can use the exception stack decoder to figure out where it is actually crashing:

https://github.com/me-no-dev/EspExceptionDecoder

notsolowki, 2023-10-06:
I WAS USING n esp32 WROOM 32 non variant super cheap mini board. it crashed like crazy, after getting the new esp32 that says espressif on the chip im not having crashes anymore. it seems to work perfect now. Note the new board claims to be esp32 wroom 32e.

this is a link to the exact one i bought
https://www.amazon.com/gp/aw/d/B0C9TGJRPH?psc=1&ref=ppx_pop_mob_b_asin_image

closed 2023-10-06 by notsolowki

notsolowki, 2023-10-06:
I change the lib like this to make it not hang

`I was using esp32 WROOM 32 non variant super cheap mini board. it crashed like crazy, after getting the new esp32 that says espressif on the chip im not having crashes anymore. it seems to work perfect now. Note the new board claims to be esp32 wroom 32e.

this is a link to the exact one i bought
https://www.amazon.com/dp/B0C9TGJRPH/ref=redir_mobile_desktop/143-8877457-1326723?_encoding=UTF8&psc=1&ref=ppx_pop_mob_b_asin_image
pins 4&5

Keep in mind i did still have to add vTaskDelay();

I change the lib like this and it worked,
`void task_LowLevelRX(void *pvParameters)
{

    ESP32CAN* espCan = (ESP32CAN*)pvParameters;
    while (1)
    {
int a=1;
if(a){
	 vTaskDelay(1);
a=0;
}
        twai_message_t message;
        if (twai_receive(&message, pdMS_TO_TICKS(100)) == ESP_OK)
        {
            espCan->processFrame(message);
        }
    }
}``