collin80/esp32_can #63: Can't compile using PlatgormIO

samarkh, 2024-01-08:
Hi 
I get a whole load of  "error: 'debuggingMode' was not declared in this scope" errors when I try to build this project using PlatformIO. 
I have no problems using Arduino 2.x.
Has anyone got any suggestions?

Yours Simon M.

klasrocket, 2024-02-21:
Hi.
I ran into the same problem. Found a workaround by adding 
`extern boolean debuggingMode;`
next to:
`extern QueueHandle_t callbackQueue;`

in the end of file: esp32_can_builtin.h
I did not investigate much further since I'm only gonna use the built in can-controller.

klasrocket, 2024-02-21:
Seemed to pass the build process but got stuck at linking..
```

Dependency Graph
|-- can_common @ 0.3.0
|-- ESP32_CAN @ 0.2.5
|-- SPI @ 2.0.0
Building in release mode
Linking .pio/build/esp32doit-devkit-v1/firmware.elf
/home/linus/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32doit-devkit-v1/lib252/libESP32_CAN.a(esp32_can_builtin.cpp.o):(.literal._ZN8ESP32CAN10_setFilterEjjb+0x0): undefined reference to `debuggingMode'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32doit-devkit-v1/firmware.elf] Error 1
```

klasrocket, 2024-02-21:
Finally got it uploaded using even uglier workaround by initiating the boolean in begining of each function were used :/