collin80/esp32_can #66: ESP32 + MCP2518FD Pro board has errors when connected to can network ** and !!

kukumagi, 2024-05-09:
ESP32 + MCP2518FD Pro board has errors when connected to can network ** and !!
Setup configures the MCP2518 correctly but when connected to can network I get !! and ** in serial monitor.

In code I found that these are printed from mcp2517fd.cpp file
if (diagBits & 0x3030000) //either NBIT0 or NBIT1 error (or DBIT0, DBIT1)
and
if (diagBits & 0x38380000) //19 - RX Fixed form, Bit stuff, or CRC error, either FD or not

But I have no idea what these really mean or how to get it to work.

kukumagi, 2024-05-14:
Yes I was testing with FD traffic.

kukumagi, 2024-05-17:
Just tested.
1. Normal can to normal can (baud 500k) - Works
2. Normal can to canfd (baud 500k and 2000k) - works
3. canfd to canfd - does not work and gives the ** and !! output.

kukumagi, 2024-05-17:
When using 500k 4000k canfd then I get !! error when the other MCP is sending

collin80, 2024-05-21:
I have actually seen this. It should be documented that, for some reason, using a CAN speed of 500k does not work with this library. If you set the CAN speed to 1M then you can set the FD speed to 2M, 4M, 5M, etc and it works fine. I don't know why this is. But, seeing the settings from ACAN could help to narrow down what is going on. So, thank you.

kukumagi, 2024-05-21:
Setting the CAN1.debuggingMode = true
And CAN1.beginFD(500000, 2000000)
Gives this output 

Diag1: 10020000     ErrFlgs: 8Diag0: 670059
**!!**!!**!!MCP2517FD SPI Inited
_initFD()
Nomimal Settings:  TSEG1: 63   TSEG2: 16
Data Settings:  TSEG1: 15   TSEG2: 4
commonInit()
11101000000011110000010010000000
10000000000100111
Mode
!!!!**!!**!!**!!**!!**!!10101
0
MCP2517 InitFD Success

Also tried 1M-2M CAN1.beginFD(1000000, 2000000)

I get this output:
Reset 2517FD hardware
     Diag1: 10000000     ErrFlgs: 8Diag0: 200000
MCP2517FD SPI Inited
_initFD()
Nomimal Settings:  TSEG1: 31   TSEG2: 8
Data Settings:  TSEG1: 15   TSEG2: 4
commonInit()
11101000000011110000010010000000
10000000000100111
Mode
10101
0
MCP2517 InitFD Success
!!!!!!!!!!!!!!!!Reset 2517FD hardware

And 1M x2 from ACAN2517FD
Configure ACAN2517FD
MCP2517FD RAM Usage: 2016 bytes
Bit Rate prescaler: 1
Arbitration Phase segment 1: 31
Arbitration Phase segment 2: 8
Arbitration SJW:8
Actual Arbitration Bit Rate: 1000000 bit/s
Exact Arbitration Bit Rate ? yes
Arbitration Sample point: 80%

vanryssel, 2024-06-06:
Hi there!

I think the problem here is the setting of SJW inside the _initFD function of the fixed value of "4", called from initFD:
nominalCfg.bF.SJW = sjw;
dataCfg.bF.SJW = sjw;

The manual says: 
Select the Largest Possible NSJW and DSJW
- Maximizes the oscillator tolerance.
- Allows the receiving nodes to quickly resynchronize to the transmitting nodes.

So i changed this to:

nominalCfg.bF.SJW = tseg2 - 1;
dataCfg.bF.SJW = tseg2 - 1;

after the calculation of tseg2.

Now the error is gone :-)

vanryssel, 2024-06-06:
Short update: 
the error was gone for two mcp2518FD devices onto the bus, after adding a third device (Teensy 4.1) the error is back :-(

So trying further...

vanryssel, 2024-06-06:
Next update:
now it works with all three nodes, the failure was a mis-configured TX pin on the Teensy side :-)
Thanx!

kukumagi, 2024-06-07:
Confirmed. Works.

> Hi there!
>
> I think the problem here is the setting of SJW inside the _initFD function of the fixed value of "4", called from initFD: nominalCfg.bF.SJW = sjw; dataCfg.bF.SJW = sjw;
>
> The manual says: [...]