collin80/esp32_can #69: Integrate this project for Adafruit RP2040 CAN bus Feather

as4mz, 2024-09-11:
My basics are not very strong on this kind of projects. Previousy, I used Macchina M2 with M2RET to get the OBD-II data from Honda car. 

Now, I am wondering is it possible to port this project for RP2040 CAN bus feather (https://learn.adafruit.com/adafruit-rp2040-can-bus-feather/overview)? RP2040 CAN bus feather is relative cheap (~20$) and I could use other boards to get the OBD-II data from RP2040 via serial inputs. 

Thanks for your help!

as4mz, 2024-09-11:
This might not be the proper medium for asking. I truely appreciate any help :)

collin80, 2024-09-17:
Well, CAN libraries do exist for that board. But, perhaps you are looking for something like ESP32RET or M2RET that actually handles all the details of using the library to create an interface that you can use for CAN capture and sending. 

In principle, sure, one of the existing *RET projects could be used as a basis for making a RET project for that board as well. I don't have any RP2040 boards yet (odd, I know). I guess I just have so many ESP32 boards and such that I haven't found any reason to use an RP2040 yet.

So, right now, I'd suggest an ESP32 based CAN board (there are LOTS). That way, ESP32RET could easily be modified to work.

as4mz, 2024-09-17:
Thanks for your reply. Can you share any links of the recommended ESP32 boards? How about this one (https://www.adafruit.com/product/5364) with transeiver (https://www.adafruit.com/product/5708)?