collin80/esp32_can #72: Correct application of filters for multiple IDs

brightproject, 2024-12-07:
Hello @collin80 🙂
Please help me.
I have two devices.
One is a transmitter, based on `stm32f411` + `mcp2515`
The code is something like this:

```
#include <mcp_can.h>

unsigned char canMsg1[8] = {0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17};
unsigned char canMsg2[8] = {0x20, 0x21, 0x22, 0x23, 0x24, 0x25, 0x26, 0x27};
unsigned char canMsg3[8] = {0x30, 0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37};

MCP_CAN CAN(PA4);

void setup() {
  CAN.begin(MCP_ANY, CAN_500KBPS, MCP_8MHZ)
  CAN.setMode(MCP_NORMAL);
}

void loop() {
CAN.sendMsgBuf(0x10, 0, 8, canMsg1); 
CAN.sendMsgBuf(0x30, 0, 8, canMsg2); 
CAN.sendMsgBuf(0x50, 0, 8, canMsg3); 
}
```

The second receiving device, based on `ESP32` + `SN65HVD230D`
The code is something like this:

```
#include <esp32_can.h>

void printFrame(CAN_FRAME &frame);

void setup() {
Serial.begin(115200);
CAN0.setCANPins(GPIO_NUM_21, GPIO_NUM_18);
if(CAN0.begin(500000))
{
Serial.println("Builtin CAN Init OK");
} else {
Serial.println("BuiltIn CAN Init Failed");
}

// CAN0.setRXFilter(0, 0x28, 0x7FF, false);
CAN0.watchFor(0x28, 0x7FF); //allow everything else through
CAN0.watchFor(); //allow everything else through
}

void loop() {
        CAN_FRAME message;

        if (CAN0.read(message)) {
            printFrame(message);
        }
}

void printFrame(CAN_FRAME &frame)
{
	Serial.print("ID: ");
	Serial.println(frame.id,HEX);
	Serial.print("Ext: ");
	if(frame.extended) {
		Serial.println("Y");
	} else {
		Serial.println("N");
	}
	Serial.print("Len: ");
	Serial.println(frame.length, DEC);
	for(int i = 0;i < frame.length; i++) {
		Serial.print(frame.data.uint8[i], HEX);
		Serial.print(" ");
	}
	Serial.println();
}
```
I can't filter messages, all messages coming to the `CAN` bus come to the receiving device and the filters don't work, or I configure them incorrectly.
The code was taken from the examples

> https://github.com/collin80/esp32_can/blob/66d1440c9e53ccaeaccb6bfa91bde89cf7e368e6/examples/CANTestESP32_Both/CANTestESP32_Both.ino#L63

and

> https://github.com/collin80/esp32_can/blob/66d1440c9e53ccaeaccb6bfa91bde89cf7e368e6/examples/CANTestESP32_BuiltIn/CANTestESP32_BuiltIn.ino#L41

collin80, 2024-12-10:
This line you've got in there lets all traffic through:

CAN0.watchFor(); //allow everything else through

Don't have that line if you want to filter. Instead just keep the line that lets 0x30 through and that's all that should get through.

brightproject, 2024-12-10:
Thank you for your response.
I found some explanation in the code, although this is a slightly different project, but I think there are some common points with `esp32_can`.

> https://github.com/collin80/due_can/blob/master/howtouse.txt