collin80/esp32_can #75: ESP32 CAN not working after update ArduinoESP32 to v3.1.1 and esp32_can to 0.3.0

jeroenveer, 2025-01-15:
After update ArduinoESP32 to v3.1.1 and esp32_can to 0.3.0 board hangs on a ESP32.

the main loop hangs after:
`xTaskCreate(ESP32CAN::task_CAN, canHandlerTaskName, 8192, this, 15, &task_CAN_handler);`

it seems task_CAN keeps running.

Same code on a ESP32-C6 works without issues.

```
#include <esp32_can.h>

void setup() {
	Serial.begin(115200);

	Serial.println("Initializing ...");

	CAN0.setDebuggingMode(true);
	CAN0.setCANPins(GPIO_NUM_4, GPIO_NUM_5);
	CAN0.begin(CAN_BPS_500K);
	CAN0.watchFor();
}

void loop() {

	CAN_FRAME message;
	if (CAN0.read(message)) {
		printFrame(&message);
		message.id = 0x100;
		CAN0.sendFrame(message);
	}
}
```

```
Port open
t to 0x400d4d40
[    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d4d10
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d4ce0
[    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d4cb0
[   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d4d40
[   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d4d10
[   134][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d4ce0
[   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d4cb0
f�(359) esp_core_dump_flash: No core dump partition found!
E (359) esp_core_dump_flash: No core dump partition found!
[   163][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   195][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb60
[   206][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb60
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.01
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   367648 B ( 359.0 KB)
  Free Bytes        :   327584 B ( 319.9 KB)
  Allocated Bytes   :    32816 B (  32.0 KB)
  Minimum Free Bytes:   322272 B ( 314.7 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  4608.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00490000, size:  4608.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00910000, size:  7104.0 KB, type: DATA, subtype: SPIFFS
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan 15 2025 15:26:12
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-282-gcfea4f7c98-dirty
  Arduino Version   : 3.1.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=16M,PartitionScheme=7mb_spiffs,DebugLevel=verbose,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============
[  4697][V][esp32-hal-uart.c:421] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  4706][V][esp32-hal-uart.c:510] uartBegin(): UART0 not installed. Starting installation
[  4717][V][esp32-hal-uart.c:575] uartBegin(): UART0 initialization done.
VMDPR_[  4743][V][esp32-hal-uart.c:421] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  4752][V][esp32-hal-uart.c:434] uartBegin(): UART0 Driver already installed.
Initializing ...
[  4759][D][esp32_can_builtin.cpp:268] init(): [CAN] Init called
Built in CAN Init
_init done
[  4767][D][esp32_can_builtin.cpp:270] init(): [CAN] Init done
Driver installed - bus 0
Creating queues
Starting can handler task
```

jeroenveer, 2025-01-15:
adding vTaskDelay(pdMS_TO_TICKS(100)); to task_CAN gets the loop to run again.
Not sure if this is a correct way to fix it and if its a good value as i took it from task_LowLevelRX.

esp32_can_builtin.cpp
```
void ESP32CAN::task_CAN( void *pvParameters )
{
    ESP32CAN* espCan = (ESP32CAN*)pvParameters;
    CAN_FRAME rxFrame;

    for (;;)
    {
        if (uxQueueMessagesWaiting(espCan->callbackQueue)) {
            //receive next CAN frame from queue and fire off the callback
            if(xQueueReceive(espCan->callbackQueue, &rxFrame, portMAX_DELAY) == pdTRUE)
            {
                espCan->sendCallback(&rxFrame);
            }
        }
        else vTaskDelay(pdMS_TO_TICKS(100));

#if defined(CONFIG_FREERTOS_UNICORE)
        delay(10);
#endif
    }

    vTaskDelete(NULL);
}
```

Modellfan, 2025-01-30:
Created a pull request on this: https://github.com/collin80/esp32_can/pull/78

The delay should be propably as low as possible. Maybe we just can reset the watchdog also manually here