collin80/esp32_can #82: arduino IDE macchina A0 board esp32-can.h crashes at CAN0.begin(5000);

weeblebiker, 2025-03-26:
arduino IDE, board definitions and libraries are up to date, tried running on 2 different A0's and several sketches and the example sketches.  nothing gets past   I was running the sketches and using this module fine until a code corruption occurred.  tested on a known working module and it hanges on the same CAN0.begin(5000); line.

tloakes, 2025-03-26:
Fix is here: https://github.com/collin80/esp32_can/issues/75

weeblebiker, 2025-03-26:
yes it's hanging at starting can handle task like #75. deleted esp32_ca-master again and downloaded the latest code again and it is still hanging at starting can handler task when it hits CAN0.begin(500000);.
so this is not fixed and I need to add a delay like in #75????

weeblebiker, 2025-03-27:
added the delay and it doesn't hang at "starting can handler task",  however, the code is now lagging. I changed the delay from 100 to 1 in the line:  else vTaskDelay(pdMS_TO_TICKS(100));   and it isn't hanging.  but a appropriate fix isn't in place.

weeblebiker, 2025-03-27:
I'm sending a can query every 90ms

collin80, 2025-03-28:
I just pushed a fix to the master branch. It delays for 100ms when first starting but then only 4ms and only if no CAN traffic was found. In practice this all seems to work OK for me. Sorry it took so long to commit it.

weeblebiker, 2025-03-28:
fixed.
Thanks Collin!

closed 2025-03-29 by weeblebiker